eszdman/PhotonCamera #7: More informations about technical which libHDRX use

unknow-dev, 2020-06-27:
Hi @eszdman, Can you share some informations about the technology applied in libHDRX.so?

Thansk.

eszdman, 2020-06-27:
Hello, we use the library on halide in order to pyramidal align and merge the images that the camera receives, this can be seen from the Wrapper.java source.
If you wish, you can replace this library with your implementation.

unknow-dev, 2020-06-27:
Thanks for your reply.
It seems you use the same approach like  https://github.com/timothybrooks/hdr-plus, right?

eszdman, 2020-06-27:
Yes, that's right, based on these alignment algorithm

unknow-dev, 2020-06-27:
I was make a comparison between your HDRX raw output and "hdr-plus" raw output (from same raws step input), I realize that your results are less noise than "hdr-plus" output.

eszdman, 2020-06-27:
Hdplus implementation is not designed for mobile dng and CPU's and requires tuning

unknow-dev, 2020-06-27:
And another issue that I see it affects both your module and hdr-plus, if there is an object moving between frames, the result have some small blocks overlap.

eszdman, 2020-06-27:
Robust model requires adjustments, this is fixed in recent dev updates

unknow-dev, 2020-06-27:
Thanks for your infomations :D
One more question, Do you have any plan to make some PR to  https://github.com/timothybrooks/hdr-plus to improve it results, that repo look like be still maintained to make it better or are your technologies is private?

eszdman, 2020-06-27:
Perhaps later the old library will be opened as a separate repository, as the main algorithms will be redone to work on GPU.

closed 2020-06-30 by assasinfil