eszdman/PhotonCamera #16: Please set versionCode

IzzySoft, 2020-09-19:
The APK attached to the current release doesn't have `versionCode` set:

    package: name='com.eszdman.photoncamera' versionCode='' versionName='0.3' compileSdkVersion='29' compileSdkVersionCodename='10'

That makes it hard to keep versions apart, as Android uses this field to determine if an APK file is an update or not. Could you please set that field [in your `build.gradle`](https://github.com/eszdman/PhotonCamera/blob/master/app/build.gradle#L42)? I see you're referring to a variable there which probably isn't set. Especially if you want your app included with [F-Droid](https://f-droid.org/) once it's "stable", a literal value in this place would be very helpful (required by F-Droid's update check process).

Thanks!

eszdman, 2020-10-07:
Is your problem fixed now? Dev branch

IzzySoft, 2020-10-07:
Checking the latest APK, it seems not:

```
package: name='com.eszdman.photoncamera' versionCode='' versionName='0.66' compileSdkVersion='30' compileSdkVersionCodename='11'
```

`versionCode` is still empty. I'm no dev and have no build env set up here, so I cannot test to build a version myself.

assasinfil, 2020-10-07:
in dev branch version code is automatically increment

closed 2020-10-07 by assasinfil

IzzySoft, 2020-10-07:
OK, thanks! Looking forward to the next release then.