eszdman/PhotonCamera #17: Problem with Exynos

oreothish, 2020-09-30:
**Problem:**
Whenever trying to take a photo, there's always the same crash

**Settings:**

[image]

**Logcat:**

```
time: 1601497451723
msg: java.lang.RuntimeException: Error creating shader.
stacktrace: java.lang.RuntimeException: Error creating shader.
	at com.eszdman.photoncamera.processing.opengl.GLProg.compileShader(GLProg.java:103)
	at com.eszdman.photoncamera.processing.opengl.GLProg.useProgram(GLProg.java:66)
	at com.eszdman.photoncamera.processing.opengl.nodes.postpipeline.GlobalTonemaping.Run(GlobalTonemaping.java:39)
	at com.eszdman.photoncamera.processing.opengl.GLBasePipeline.runAll(GLBasePipeline.java:51)
	at com.eszdman.photoncamera.processing.opengl.nodes.postpipeline.PostPipeline.Run(PostPipeline.java:117)
	at com.eszdman.photoncamera.processing.ImageProcessing.ApplyHdrX(ImageProcessing.java:465)
	at com.eszdman.photoncamera.processing.ImageProcessing.Run(ImageProcessing.java:487)
	at com.eszdman.photoncamera.processing.ImageSaver.done(ImageSaver.java:83)
	at com.eszdman.photoncamera.processing.ImageSaver.Process(ImageSaver.java:217)
	at com.eszdman.photoncamera.processing.ImageSaver.lambda$run$0$ImageSaver(ImageSaver.java:112)
	at com.eszdman.photoncamera.processing.-$$Lambda$ImageSaver$z_xT8ahTvCohpjCEjOl3obH29jU.handleMessage(Unknown Source:2)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.loop(Looper.java:214)
	at android.os.HandlerThread.run(HandlerThread.java:67)
```

**Device Specs:**

- Device : Samsung Galaxy J6 - 2 GB Ram Variant
- Board: Exynos 7870
- Rom: CrDroid 6.9 - Android Q

snajdovski, 2020-10-01:
Same on my s7 Edge I can confirm (Android 8 Oreo Stock), also Android 10 (FloydROM OneUI 2.1)

eszdman, 2020-10-03:
Fixed

closed 2020-10-05 by eszdman