eszdman/PhotonCamera #22: Actions - sign apk

assasinfil, 2020-10-07:
Need fix apk sign in actions.

eszdman, 2020-10-08:
Fix it

rancidfrog, 2020-11-09:
Do debug builds in releases need to be uninstalled before updating to new versions?
As I am unable to install latest release:
[image]

Signed APKs would be great.

snajdovski, 2020-11-09:
The app is under rapid development and signing apks is not on top of our priority list, simple reinstall (Uninstall old Photon build and install again).

rancidfrog, 2020-11-09:
But it can be automatic and rapid.
This project is no where near as rapid as https://github.com/M66B/FairEmail/commits/master
and yet developer has time to sign each and every apk, many times more than one or two a day. So that should not be an excuse.

Also,
there is no backup and restore settings to make this _rapid_  transition swift and easy

snajdovski, 2020-11-10:
> But it can be automatic and rapid.
> This project is no where near as rapid as https://github.com/M66B/FairEmail/commits/master
> and yet developer has time to sign each and every apk, many times more [...]

Lmao do you think dislikes will make me feel bad?
It could be done I just said its not on the top of the prio list. And you started yik yaking around with dislikes.

snajdovski, 2020-11-10:
> But it can be automatic and rapid.
> This project is no where near as rapid as https://github.com/M66B/FairEmail/commits/master
> and yet developer has time to sign each and every apk, many times more [...]

Open new issue for that also.

rancidfrog, 2020-11-10:
Yik yaking. I just mentioned another project where the developer is dedicated and if you look at commit history you will see his work ethic. 
As you said it is not a priority, but nothing to do with rapid developments.
Sorry for any misunderstanding.

**<ins>Suggestions</ins>** :

- Sign and upload compiled apk to Github releases automatically using Travis CI: https://gist.github.com/mariotaku/7a0c51955d14def2fa0e
- Use Travis to build and save an APK: https://gist.github.com/domenic/ec8b0fc8ab45f39403dd

assasinfil, 2020-11-10:
> Yik yaking. I just mentioned another project where the developer is dedicated and if you look at commit history you will see his work ethic.
> As you said it is not a priority, but nothing to do with rapid developments.
> Sorry for any misunderstanding.
> 
> **Suggestions** :
> 
> * Sign and upload compiled apk to Github releases automatically using Travis CI: https://gist.github.com/mariotaku/7a0c51955d14def2fa0e
> * Use Travis to build and save an APK: https://gist.github.com/domenic/ec8b0fc8ab45f39403dd

We use github actions to build and sign APKs.
But problem in actions script.

snajdovski, 2020-11-10:
> Yik yaking. I just mentioned another project where the developer is dedicated and if you look at commit history you will see his work ethic.
> As you said it is not a priority, but nothing to do with [...]

You want to say that devs here are not dedicated to their work? And we dont have work ethic, sorry for lowering your expectations and wasting 5 sec of uninstalling an old apk build and installing new one, since that is very hard for you I suggest using propratery software from play store that are 'signed' by whole mighty Google Saints.@eszdman works hard on this project and he is probably very busy, if you are that knowledgeable you can contribute with signing apk's.
You all know how to report issues that are basically finding a hair in an egg, but don't want to contribute.

I don't think that signing an .apk is big of a deal to argue about, because first its not necessary, it may be anoying but that is how it is, key.jks files are vulnerable things to upload to github, as GitHub script kiddies could steal it and sign apks and upload to the whole mighty goolag.

rancidfrog, 2020-11-10:
😂🤣
I see.
Feeling quite social in isolation, or something?
I was just giving an example where dev was **also**  on rapid release cycle and yet you do npt need to uninstall and reinstall app each time.
Yes, it is a slight annoyance, but an annoyance nonetheless.
I was not angry, as you clearly are.
You just like demeaning people for no reason, when you just coupd have ignored.
Whereas, @assasinfil just gave a quick response  to my comment, without making excuses or anything.
You should stop wasting your time with comments, maybe take up boxing to release all that stress, anger, or whatever thingamajiggy is bothering you
😂🤣

snajdovski, 2020-11-10:
> 😂🤣
> I see.
> Feeling quite social in isolation, or something?
> I was just giving an example where dev was **also** on rapid release cycle and yet you do not need to uninstall and reinstall app each time.
> Yes, it is a slight annoyance, but an annoyance nonetheless.
> I was not angry, as you clearly are.
> You just like demeaning people for no reason, when you just could have ignored.
> Whereas, @assasinfil just gave a quick response to my comment, without making excuses or anything. Thank you for clarifying.
> You should stop wasting your time with comments, maybe take up boxing to release all that stress, anger, or whatever thingamajiggy is bothering you
> 😂🤣

You are right I should start boxing people or start drinking.