eszdman/PhotonCamera #23: Share method improvements

rancidfrog, 2020-10-20:
When sharing from photon app after taking a photograph, the multiple share optiins do not show:

Photon app share implementation:
[image]

The share is general, recent optiims not shown

Default camera app, and other apps as well, share implementation:
[image]

Not only can you share by apps, but recent options, i.e. whatsapp/instagram/other apps, are included

eszdman, 2020-10-30:
Try using this type of share
I am using android 10 and recent applications working, most frequent applications on top

closed 2021-01-12 by vibhorSrv

rancidfrog, 2021-01-12:
@vibhorSrv
Has anything been changed?
In recent android versions there is the ability to share not only with most used apps, but also with most recent sent contact. Which was what I was referring to, this way you can send directly to contact or group without selecting app and then contact.

vibhorSrv, 2021-01-12:
what you are referring to is achieved by using "android.intent.action.CHOOSER". and yes it has been implemented  recently.
https://git.io/JtvFV