eszdman/PhotonCamera #27: Input Bytebuffer of HDRX library.

unknow-dev, 2020-10-30:
Hi @eszdman,

Im trying to use a bytebuffer which read from a *.dng for input of HDRX instead of bytebuffer from Image but result is always black.

I would like to ask what is correctly type of bytebuffer for HDRX lib?

Thanks.

eszdman, 2020-10-30:
You need a direct bytebuffer
Otherwise it won't work.

unknow-dev, 2020-10-30:
Thanks for quick reply.
As you comment, i use allocateDirect to create direct bytebuffer as below, but it still not work.

public static ByteBuffer readBytesFromFile(String filePath) {

        FileInputStream fileInputStream = null;
        byte[] bytesArray = null;

        try {

            File file = new File(filePath);
            bytesArray = new byte[(int) file.length()];
            fileInputStream = new FileInputStream(file);
            fileInputStream.read(bytesArray);

        } catch (IOException e) {
            e.printStackTrace();
        } finally {
            if (fileInputStream != null) {
                try {
                    fileInputStream.close();
                } catch (IOException e) {
                    e.printStackTrace();
                }
            }

        }
        ByteBuffer byteBufferOut = ByteBuffer.allocateDirect(bytesArray.length);
        byteBufferOut.position(0);
        byteBufferOut.put(bytesArray);
        byteBufferOut.position(0);
        return byteBufferOut;

    }

eszdman, 2020-10-30:
You may have entered the wrong number of frames or entered less buffers, also input only bayer16 without packing, not usual .dng files

KillerInk, 2020-11-04:
using a dng direct wont work. you are reading the full dng including the metadata, but you only need the data.

unknow-dev, 2020-11-05:
@KillerInk  @eszdman 
I was made a quick test, write direct bytebuffer from Image and read it back.

public static File writeDirect(Context context, Image image) {
        ByteBuffer buffer;
        try {
            buffer = image.getPlanes()[0].getBuffer();
        } catch (IllegalStateException e) {
            return null;
        }
        if (buffer != null) {
            File tmpDataFile;
            try {
                tmpDataFile = File.createTempFile(
                        "RAW",  /* prefix */
                        ".raw",         /* suffix */
                        context.getExternalFilesDir("RAW")      /* directory */
                );
                FileChannel wChannel = new FileOutputStream(tmpDataFile, false).getChannel();

                wChannel.write(buffer);

                wChannel.close();
            } catch (IOException e) {
                e.printStackTrace();
                return null;
            } finally {
                image.close();
                buffer.clear();
            }
            return tmpDataFile;
        }
        return null;
    }

But it still not work.

KillerInk, 2020-11-05:
for me this code looks correct. but didnt test it.
sure the image format is RAW_SENSOR?

unknow-dev, 2020-11-05:
> sure the image format is RAW_SENSOR?

Yes, in my test, write ByteBuffer from:
https://github.com/eszdman/PhotonCamera/blob/ce8311a516fe4b3b9bafefb66f9e2bde5a195cb5/app/src/main/java/com/eszdman/photoncamera/processing/ImageProcessing.java#L218

and read it back after.

eszdman, 2021-09-18:
Should be fixed in current dev branch

closed 2021-09-18 by eszdman