eszdman/PhotonCamera #34: Latest release crashes on launch

rancidfrog, 2020-12-17:
Version: PhotonCamera-0.702515-debug
Crash on launch,

```
12-17 17:36:20.158  1312  1337 I ActivityTaskManager: START u0 {cmp=com.eszdman.photoncamera/.ui.camera.CameraActivity} from uid 10170
12-17 17:36:20.174  2168  2168 D ImageWallpaper: wallpaper visibility changes to: false
12-17 17:36:20.183 15360 15360 W ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@79634ee
12-17 17:36:20.189   866   866 E Layer   : [Surface(name=AppWindowToken{63c8eb5 token=Token{291aec ActivityRecord{4f1e99f u0 com.eszdman.photoncamera/.ui.SplashActivity t4638}}})/@0xc150569 - animation-leash#0] No local sync point found
12-17 17:36:20.189   866   866 E Layer   : [Surface(name=AppWindowToken{63c8eb5 token=Token{291aec ActivityRecord{4f1e99f u0 com.eszdman.photoncamera/.ui.SplashActivity t4638}}})/@0xc150569 - animation-leash#0] No local sync point found
12-17 17:36:20.189   866   866 E Layer   : [Surface(name=AppWindowToken{b413f75 token=Token{40316ac ActivityRecord{1cb7c5f u0 com.android.launcher3/.lineage.LineageLauncher t2989}}})/@0xc58498 - animation-leash#0] No local sync point found
12-17 17:36:20.189   866   866 E Layer   : [Surface(name=AppWindowToken{b413f75 token=Token{40316ac ActivityRecord{1cb7c5f u0 com.android.launcher3/.lineage.LineageLauncher t2989}}})/@0xc58498 - animation-leash#0] No local sync point found
12-17 17:36:20.199 15360 15360 E an.photoncamer: Invalid ID 0x00000000.
12-17 17:36:20.246 15360 15360 D AndroidRuntime: Shutting down VM
12-17 17:36:20.247 15360 15360 E AndroidRuntime: FATAL EXCEPTION: main
12-17 17:36:20.247 15360 15360 E AndroidRuntime: Process: com.eszdman.photoncamera, PID: 15360
12-17 17:36:20.247 15360 15360 E AndroidRuntime: java.lang.RuntimeException: Unable to start activity ComponentInfo{com.eszdman.photoncamera/com.eszdman.photoncamera.ui.camera.CameraActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'java.lang.Object[] java.util.Set.toArray(java.lang.Object[])' on a null object reference
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3270)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3409)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:214)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7356)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'java.lang.Object[] java.util.Set.toArray(java.lang.Object[])' on a null object reference
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at com.eszdman.photoncamera.api.CameraManager2.init(CameraManager2.java:54)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at com.eszdman.photoncamera.api.CameraManager2.<init>(CameraManager2.java:42)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at com.eszdman.photoncamera.settings.PreferenceKeys.setDefaults(PreferenceKeys.java:50)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at com.eszdman.photoncamera.ui.camera.CameraActivity.onCreate(CameraActivity.java:41)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.Activity.performCreate(Activity.java:7824)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.Activity.performCreate(Activity.java:7813)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1306)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3245)
12-17 17:36:20.247 15360 15360 E AndroidRuntime: 	... 11 more
12-17 17:36:20.248  1312 18865 I DropBoxManagerService: add tag=data_app_crash isTagEnabled=true flags=0x2
12-17 17:36:20.249  1312  3220 W ActivityTaskManager:   Force finishing activity com.eszdman.photoncamera/.ui.camera.CameraActivity
12-17 17:36:20.256 15360 15360 I Process : Sending signal. PID: 15360 SIG: 9
12-17 17:36:20.268  1312  1337 I ActivityManager: Process com.eszdman.photoncamera (pid 15360) has died: fore TOP 
12-17 17:36:20.268  1312  1379 I libprocessgroup: Successfully killed process cgroup uid 10170 pid 15360 in 0ms
12-17 17:36:20.269   663   663 I Zygote  : Process 15360 exited due to signal 9 (Killed)
12-17 17:36:20.277  1312  1369 W WindowManager: Failed looking up window callers=com.android.server.wm.WindowManagerService.windowForClientLocked:5318 com.android.server.wm.WindowManagerService.removeWindow:1782 com.android.server.wm.Session.remove:176 
12-17 17:36:20.297  2168  2168 D ImageWallpaper: wallpaper visibility changes to: true
12-17 17:36:20.750  1312  1368 W ActivityTaskManager: Activity top resumed state loss timeout for ActivityRecord{e2ef8a1 u0 com.eszdman.photoncamera/.ui.camera.CameraActivity t-1 f}
12-17 17:36:20.755   949  1047 I ThermalEngine: vs_get_temperature: read[0] xo_therm 29862 mC, weight[0] 2
12-17 17:36:20.756   949  1047 I ThermalEngine: vs_get_temperature: read[1] quiet_therm 28899 mC, weight[1] 1
12-17 17:36:21.155  1312  1910 I ActivityTaskManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 cmp=com.eszdman.photoncamera/.ui.SplashActivity bnds=[437,692][643,1005]} from uid 10100
12-17 17:36:21.172   821   842 E ANDR-PERF-UTIL: Failed to update node /sys/class/mmc_host/mmc0/clk_scaling/enable with value 0
12-17 17:36:21.172   821   842 E ANDR-PERF-RESOURCEQS: Failed to apply optimization [11, 4]
12-17 17:36:21.178  1312  1378 I ActivityManager: Start proc 15361:com.eszdman.photoncamera/u0a170 for pre-top-activity {com.eszdman.photoncamera/com.eszdman.photoncamera.ui.SplashActivity}
12-17 17:36:21.181 15361 15361 I an.photoncamer: Late-enabling -Xcheck:jni
12-17 17:36:21.196 15361 15361 E an.photoncamer: Unknown bits set in runtime_flags: 0x8000
12-17 17:36:21.618 15361 18887 D native  : Found attached 0
12-17 17:36:21.618 15361 18887 D native  : Calling class is: java.lang.Class
12-17 17:36:21.618 15361 18888 D native  : Found attached 0
12-17 17:36:21.618 15361 18888 D native  : Calling class is: java.lang.Class
12-17 17:36:21.676  1312  2939 I ActivityTaskManager: START u0 {cmp=com.eszdman.photoncamera/.ui.camera.CameraActivity} from uid 10170
12-17 17:36:21.699 15361 15361 W ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@79634ee
12-17 17:36:21.711  2168  2168 D ImageWallpaper: wallpaper visibility changes to: false
12-17 17:36:21.714 15361 15361 E an.photoncamer: Invalid ID 0x00000000.
12-17 17:36:21.727   866   866 E Layer   : [Surface(name=AppWindowToken{e292b68 token=Token{bb63e8b ActivityRecord{33aec5a u0 com.eszdman.photoncamera/.ui.SplashActivity t4639}}})/@0xd6993ac - animation-leash#0] No local sync point found
12-17 17:36:21.727   866   866 E Layer   : [Surface(name=AppWindowToken{b413f75 token=Token{40316ac ActivityRecord{1cb7c5f u0 com.android.launcher3/.lineage.LineageLauncher t2989}}})/@0xc58498 - animation-leash#0] No local sync point found
12-17 17:36:21.758   949  1047 I ThermalEngine: vs_get_temperature: read[0] xo_therm 30000 mC, weight[0] 2
12-17 17:36:21.759   949  1047 I ThermalEngine: vs_get_temperature: read[1] quiet_therm 28899 mC, weight[1] 1
12-17 17:36:21.760 15361 15361 D AndroidRuntime: Shutting down VM
12-17 17:36:21.761 15361 15361 E AndroidRuntime: FATAL EXCEPTION: main
[... cut ...]
12-17 17:36:23.654  1312  5046 D OpenGLRenderer: endAllActiveAnimators on 0x7006ef0500 (RippleDrawable) with handle 0x70ca69f2e0
12-17 17:36:23.667  1312  2939 I ActivityManager: Killing 15361:com.eszdman.photoncamera/u0a170 (adj 900): crash
12-17 17:36:23.671  1312  1337 E InputDispatcher: Window handle Window{b00d436 u0 Application Error: com.eszdman.photoncamera} has no registered input channel
12-17 17:36:23.709   663   663 I Zygote  : Process 15361 exited due to signal 9 (Killed)
12-17 17:36:23.716  1312  1379 I libprocessgroup: Successfully killed process cgroup uid 10170 pid 15361 in 48ms

```

eszdman, 2020-12-21:
Clear app data after installing

closed 2021-01-09 by eszdman

rancidfrog, 2021-01-12:
Thank you.

https://github.com/eszdman/PhotonCamera/commit/d716c3b827d48f5e2b960e0093f5bc61ef0245bd

Does this commit fix the need to clear settings on update?

eszdman, 2021-01-13:
In any case, it is recommended to reset the settings