eszdman/PhotonCamera #45: OP5: Unsupported device and crashing

mizzunet, 2021-04-11:
**Describe the bug**
Photon camera is crashing on OnePlus 5 also shows a toast that it's unsupported device. May I know what are it's requirements?

**To Reproduce**
Steps to reproduce the behavior:
1. Open Photon Camera
2. Capture photo from either front/back camera
3. See it's crashing on OnePlus 5

**Expected behavior**
Should save photo rather than ctashing.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
https://nekobin.com/boyagufoga

**Smartphone (please complete the following information):**
 - Device: OnePlus 5
 - OS: Android 11
 - Browser: Bromite(wait why?)
 - Version: Version 0.72

**Additional context**
Add any other context about the problem here.

VildaTech, 2021-04-11:
PhotonCamera have issues with following things.

Android 11
You need camera2api enable
Devices on Android 10, 9... have works much better.

I have OnePlus 6 with stock OxygenOS based on Android 10 and PhotonCamera works.

mizzunet, 2021-04-11:
> PhotonCamera have issues with following things.
> 
> Android 11
> You need camera2api enable
> Devices on Android 10, 9... have works much better.
> 
> I have OnePlus 6 with stock OxygenOS based on Android 10 and PhotonCamera works.

Camera2API has already been enabled, I guess, as GCam works fine. 
[image]
[image]

Any chance for some workaround for PhotoCamera working Android 11?

VildaTech, 2021-04-11:
Probably in the future, but PhotonCamera on Android 11 have issues.
You can try GPU align method, in some cases could help.

eszdman, 2021-09-18:
Fixed

closed 2021-09-18 by eszdman