eszdman/PhotonCamera #46: release on fdroid

c33s, 2021-04-11:
https://f-droid.org/en/contribute/

carallo, 2021-12-19:
The app can't be released on F-Droid?

licaon-kter, 2022-03-24:
Where do the libs come from? Where are their sources?

https://github.com/eszdman/PhotonCamera/app/src/main/jniLibs

eszdman, 2022-03-25:
> Where do the libs come from? Where are their sources?
> 
> https://github.com/eszdman/PhotonCamera/app/src/main/jniLibs

Freedcam eszd_hdr branch

licaon-kter, 2022-03-25:
> Freedcam eszd_hdr branch

Umm? https://github.com/KillerInk/FreeDcam/branches

eszdman, 2022-03-25:
> Umm

Yes repo is right, probably he removed this branch, I will write to him to return it. I do not add to my repository due to the fact that there may be problems with the build on different systems

okias, 2022-09-19:
Hello! This camera looks amazing. I would like to try it!

Is there currently any blocker (proprietary dependencies or build not working correctly?) or has anyone tried to get it in yet?

eszdman, 2022-09-20:
> Hello! This camera looks amazing. I would like to try it!
> 
> Is there currently any blocker (proprietary dependencies or build not working correctly?) or has anyone tried to get it in yet?

Only 1 library is critical - libaiphoto(app package name whitelist)
libhdrx not always used(it's cpu alignment, gpu is fine, fully java side)

TheModernWizard, 2023-07-30:
Is this problematic library used anymore? It's not in the list on the readme. If not, I'd love to see this app on f-droid! Its UI looks lightyears better than the other options there.

pabloscloud, 2023-11-01:
This app is on fdroid by now

licaon-kter, 2023-11-01:
@pabloscloud no, sadly not (yet)

pabloscloud, 2023-11-01:
> @pabloscloud no, sadly not (yet)

[image]

okias, 2023-11-01:
Thank you to the kind person who did the hard work! :)

licaon-kter, 2023-11-01:
@pabloscloud the topic is "in fdroid" not "binary apk from github hosted in an fdroid repo" (Izzy?)

TheModernWizard, 2023-11-01:
@licaon-kter is correct, Photon Camera is now on IzzyOnDroid, but not F-Droid itself. Helpful for sure, but if it's put on F-Droid it'll be even better.

pabloscloud, 2023-11-03:
I still think it's weird that it is referred to the official fdroid store as the while fdroid as if other repos didn't exist, but ok it's no in the official fdroid store I guess

licaon-kter, 2023-11-03:
@pabloscloud it's not strange at all, it says _"in fdroid"_ meaning _"built from the FOSS source and hosted on f-droid.org"_ and not _"blindly hosting a binary built by someone else and add that to my repos"_

pabloscloud, 2023-11-03:
I know it is that way but I personally find it weird. It's just that I open an F-Droid client and see more apps than those in the F-Droid repo so for me they are "in F-Droid" even tho they are not in their repository. But I see why I am wrong 🙂

zecamigo, 2024-12-07:
@licaon-kter Even on IzzyOnDroid apparently app is set to debuggable. :/

eszdman, 2025-01-23:
> Is this problematic library used anymore? It's not in the list on the readme. If not, I'd love to see this app on f-droid! Its UI looks lightyears better than the other camera options there.

I made a special branch for FDroid versions without prebuilt libraries 
@licaon-kter you can check now

licaon-kter, 2025-01-23:
I still see https://github.com/skgmn/RapidDecoder is pulled from a custom maven repo

we'll need to rebuild those libs I guess?

that 11 years (!!) old lib is the best you can use?

eszdman, 2025-01-23:
> I still see https://github.com/skgmn/RapidDecoder is pulled from a custom maven repo

It's not used, you also don't need to rebuild anything, check latest commit

licaon-kter, 2025-01-24:
@eszdman it builds now

ok, so, why does it connect to Github?

also, it does not take any pics 😒 

the frame count thing in the button remains as 5 or 2 at random

but appears to take videos 🤷

eszdman, 2025-01-24:
> @eszdman it builds now
> 
> ok, so, why does it connect to Github?
> 
> also, it does not take any pics 😒 
> 
> the frame count thing in the button remains as 5 or 2 at random
> 
> but appears to take videos 🤷 

Connection to GitHub is necessary to update text configs, for shooting support of RAW is required, the number of frames depends on the lighting, in the dark there are more frames

licaon-kter, 2025-01-24:
> Connection to GitHub is necessary to update text configs,

so we'd need to add [NonFreeNet](https://f-droid.org/docs/Anti-Features/#NonFreeNet) antifeature

> for shooting support of RAW is required, 

and the app can't detect if the device has this or not?

> the number of frames depends on the lighting, in the dark there are more frames

sure, but why does it remain at 2 or 5 or whatever after shooting?

licaon-kter, 2025-01-24:
> https://github.com/eszdman/PhotonCamera/blob/dev/app/SupportedList.txt

my device appears in this list, so, it has what is needed, right?

> https://github.com/eszdman/PhotonCamera/tree/dev/app/specific

these are packaged with the app too?

eszdman, 2025-01-24:
> > https://github.com/eszdman/PhotonCamera/blob/dev/app/SupportedList.txt
> 
> my device appears in this list, so, it has what is needed, right?
> 
> > https://github.com/eszdman/PhotonCamera/tree/dev/app/specific
> 
> these are packaged with the app too?

The list shows whether there is calibration data for shooting and does not always guarantee full functionality, for example when the required blobs are removed in the custom firmware. 
This data is not included in the application, but it is possible to make it as assets

licaon-kter, 2025-01-24:
> for example when the required blobs are removed in the custom firmware.

I have the official Xiaomi camera in this custom ROM, just a data point

> This data is not included in the application, but it is possible to make it as assets

that would remove the need for an anti-feature flag, yes (eg. maybe have a "update config" button that does if the user chooses?

> Perhaps you can't save images because you don't have permission to write files, 

I've granted it all files access as requested

> you can look at DCIM/PhotonCamera/PhotonLog

saw the file, will take a closer look

RokeJulianLockhart, 2025-01-24:
> that would remove the need for an anti-feature flag, yes (eg. maybe have a "update config" button that does an update if the user chooses?)

@licaon-kter, I know that I would rather have the configurations auto-update, and consequently retain the anti-feature flag.

That's because I can't imagine that many users are going to bother to manually update the configuration (considering that from their perspective it should do it itself) so they'll be using out-of-date configurations.

Really, the best solution would just be to host the files somewhere "free" – not GitHub – but I suppose moving the project even to GitLab instead is unrealistic.