eszdman/PhotonCamera #54: Crash on startup because of updateGalleryThumb() and 5k+ images in DCIM directory

frikisama, 2021-09-14:
**Describe the bug**
With LineageOS 18.1 (Android 11) on a OnePlus 7 Pro, when there is a great amount of pictures in the DCIM directory (5k+ in my case), the application crashes on startup. This is caused by the application scanning ALL image files in that directory on startup to create the thumbnails for the gallery while building the Camera fragment.

The `cameraFragmentViewModel.updateGalleryThumb();` call in the `onResume()` method in `src/main/java/com/particlesdevs/photoncamera/ui/camera/CameraFragment.java` takes several minutes in this case, and the application crashes after finishing.

**To Reproduce**
Steps to reproduce the behavior:

0. Have a sizable amount of pictures on a LineageOS 18.1 / Android 11 phone
1. Launch Photon Camera
2. Wait
3. Wait...
4. Crash

**Expected behavior**
The application starts up. Maybe first boot takes longer. Maybe gallery startup takes longer.

**Logs**
https://gist.github.com/frikisama/ff78255ff0a6ca731f705ed2c04df9e4

**Smartphone (please complete the following information):**
 - Device: OnePlus 7 Pro
 - OS: LineageOS
 - Version: 18.1 - latest build

**Additional context**
The background could be the slow FS access on Android 11. See here for example: https://github.com/syncthing/syncthing-android/issues/1630
Or here: https://www.reddit.com/r/androiddev/comments/kpn68k/android_11_very_slow_file_access_performance/

Bypassing the scanning on startup lets the application start, but it crashes getting into the gallery. Putting this in a background thread could help. Maybe a cache could help with the UX side of things, in this case. An option to use an external gallery app could be used as a workaround.

frikisama, 2021-09-15:
This patch works around this issue: https://gist.github.com/frikisama/ff736e69be0a8732735fc34477ab25a8

Work needs to be done around gallery image loading and thumbnailing, as it reads the whole directory every time, but the application itself can start even with a high amount of images in the DCIM directory on Android 11, and simply becomes unbearably slow when working with the gallery.

eszdman, 2021-09-16:
Thanks for the bug report, I will take into account that there is such a bug.

closed 2021-09-18 by eszdman

OliverLew, 2021-11-29:
Just want to add my experience, also wondering if it's solved now (in next release).

I had 2000+ images in DCIM folder, and it uses 30 seconds to 1 minute to load, also really long to open the built-in gallery or come back to the camera view from gallery or settings. Guess it's less than 5000+ and that's the reason it managed to open rather than crash?

Then I emptied the DCIM folder, there might be 200+ images somewhere else in my phone, PhotonCamera still takes ~5 seconds to open, but quite faster (< 1 second) to navigate within the app.

The phone is Oneplus 8. OS is LineageOS 18.1. PhotonCamera is v0.8.12381.

Not sure if all the slow responding is due to the same reason as the issue presented by @frikisama. Anyway I really want to use the app because it takes the best pictures for me, but the issue is preventing me from using it.

OliverLew, 2022-05-22:
I went through a quite bit to build the latest dev branch (reporting v0.8.14951), and confirm it's not freezing or crashing for me!