eszdman/PhotonCamera #74: Manual Shutter Speed increase for night/ adtrophotography

NaveenUpadhyaya, 2022-03-13:
**Is your feature request related to a problem? Please describe.**
I have a Samsung Note 8 Snapdragon and have been researching the Samsung stock camera app and other apps for astrophotography. This app looks promising, but the maximum shutter speed for Night as well as Photo modes is only 1/6 second. This is too low for low noise astrophotography. 

**Describe the solution you'd like**
Note8 camera max. Shutter speed is 10seconds, so I would like you to increase the manual shutter speed capability to 10 seconds, which will significantly improve the picture quality, in conjunction with stacking.

**Describe alternatives you've considered**
1. Increasing ISO increases the noise significantly, so it is better to increase the shutter speed.
2.  I have experimented with the noise reduction settings, but the noise is still very high owing to very high ISO (6239 in the sky picture I took with partial moon). Result will be even worse with new moon ( dark sky).

**Additional context**

eszdman, 2022-03-22:
Maximal exposure time depends on your camera 2 api parameters. 
I can only make artifical exposure time using averaging algorithms in this case

NaveenUpadhyaya, 2022-03-22:
See Camera2 API report: https://s.airbeat.com/s/XP4a

NaveenUpadhyaya, 2022-03-22:
Manual exposure time is 1/6 second max. even if I set ISO manually to 50. Some GCam versions were able to increase exposure time, but stacking feature didn't work (probably needs rooting?), so it was a dead end. Photon Camera stacking seems to work, so I am hoping it can achieve extended exposure as well.

NaveenUpadhyaya, 2022-03-25:
Actually ISO can be set manually, but it doesn't impact the maximum exposure time.