eszdman/PhotonCamera #75: Pictures regularly save as plain black images

connervieira, 2022-03-21:
**Describe the bug**
On my OnePlus 7 Pro running LineageOS, I've found that pictures taken with Photon Camera will often be saved as just plain black images. After taking an image, they'll typically appear in the built-in photo viewer just fine, then randomly become black, with the data erased. It's difficult to tell exactly what causes it. Sometimes the images will save as black immediately after taking them, while other times they'll stay normal through several app/phone restarts, only to be erased later.

**To Reproduce**
It seems difficult to reliably reproduce. I'd estimate that over half of the images I take end up getting erased/blacked-out later, but there doesn't seem to be any one thing that causes it. Here are some things I've tried to do to reproduce it, and have failed to do so consistently.
1. Take several images back to back
2. Take another image before the previous image has been processed
3. Take images in both RAW and standard image formats
4. Take images on a particular camera (telephoto, wide, etc.)
5. Enable/disable various image processing
6. Change the back-end for stacking image frames (CPU, GPU, etc.)

**Expected behavior**
After an image has been processed, it would make sense that Photon Camera shouldn't write to it again, unless explicitly asked too.

**Screenshots**
Here's several images shown in the Photon photo viewer, where random images have been overwritten with all black. These images have extremely small file sizes, and increasing the exposure shows that there is no data stored in them. Immediately after being taken, these images looked normal, then were blacked out at a later date. Despite not being edited or otherwise modified.
[image]

**Smartphone:**
 - Device: OnePlus 7 Pro
 - OS: LineageOS 18

connervieira, 2022-03-23:
This is the latest release on GitHub, V0.8

vehosah, 2025-07-23:
> What photoncamera version is used?

Latest?