eszdman/PhotonCamera #80: stops working with toast "Session onConfigFailed" on Tecno POVA

abdulhakam, 2022-05-22:
**Describe the bug**
when i capture photo, a toast appears saying Session onConfigureFailed. and image is not captured, btw on video mode, app crashes after pressing stop button.'

**To Reproduce**
Steps to reproduce the behavior:
1. Open the app
2. Press capture button in any mode.
3. See the toast.

**Expected behavior**
An image is captured and saved on the device. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Logs**
i dont know where it is saved.

**Smartphone (please complete the following information):**
 - Device: Tecno POVA (LD7)
 - OS: Android 10
 - Browser: chrome
 - Version 101

**Additional context**
i have tried many photoncamera versions from github releases, and i am 'currently trying 0.8.14660 debug version.

eszdman, 2022-09-27:
Required additional information like "camera2 test" result

abdulhakam, 2022-10-02:
<h2>Camera 0 (BACK)</h2>

<p class="small">
<strong>com.mediatek.streamingfeature:</strong>
<br>
- availableRecordStates = [0,1]
</p>

<p class="small">
<strong>android.tonemap:</strong>
<br>
- availableToneMapModes = [0,1,2]
<br>
- maxCurvePoints = 101
</p>

<p class="small">
<strong>com.mediatek.nrfeature:</strong>
<br>
- available3dnrmodes = [0,1]
</p>

<p class="small">
<strong>com.mediatek.hdrfeature:</strong>
<br>
- availableHdrModesPhoto = [0]
<br>
- availableHdrModesVideo = [0]
<br>
- availableMStreamHdrModes = [0]
</p>

<p class="small">
<strong>android.sensor:</strong>
<br>
- availableTestPatternModes = [0]
<br>
- blackLevelPattern = BlackLevelPattern([64, 64], [64, 64])
<br>
- calibrationTransform1 = ColorSpaceTransform([65536/65536, 0/65536, 0/65536], [0/65536, 65536/65536, 0/65536], [0/65536, 0/65536, 65536/65536])
<br>
- calibrationTransform2 = ColorSpaceTransform([65536/65536, 0/65536, 0/65536], [0/65536, 65536/65536, 0/65536], [0/65536, 0/65536, 65536/65536])
<br>
- colorTransform1 = ColorSpaceTransform([41877/65536, -5555/65536, -7638/65536], [-43748/65536, 102390/65536, 4342/65536], [-17786/65536, 29998/65536, 34039/65536])
<br>
- colorTransform2 = ColorSpaceTransform([94552/65536, -23292/65536, -18344/65536], [-22893/65536, 95544/65536, -7575/65536], [-4527/65536, 20534/65536, 10752/65536])
<br>
- forwardMatrix1 = ColorSpaceTransform([46483/65536, 1520/65536, 15187/65536], [19650/65536, 43222/65536, 2663/65536], [4921/65536, -28376/65536, 77536/65536])
<br>
- forwardMatrix2 = ColorSpaceTransform([35950/65536, -764/65536, 28004/65536], [9029/65536, 38585/65536, 17921/65536], [-1006/65536, -55339/65536, 110427/65536])
<br>
- maxAnalogSensitivity = 240
<br>
- orientation = 90
<br>
- referenceIlluminant1 = 21
<br>
- referenceIlluminant2 = 17
</p>

<p class="small">
<strong>android.sync:</strong>
<br>
- maxLatency = SYNC_MAX_LATENCY_PER_FRAME_CONTROL
</p>

<p class="small">
<strong>android.lens.info:</strong>
<br>
- availableApertures = [1.85]
[... 1720 lines cut ...]
</p>

<p class="small">
<strong>android.sensor.info:</strong>
<br>
- activeArraySize = Rect(0, 0 - 4208, 3120)
<br>
- colorFilterArrangement = 1
<br>
- exposureTimeRange = [100000, 400000000]
<br>
- maxFrameDuration = 400000000
<br>
- physicalSize = 4.71x3.49
<br>
- pixelArraySize = 4208x3120
<br>
- preCorrectionActiveArraySize = Rect(0, 0 - 4208, 3120)
<br>
- sensitivityRange = [100, 6400]
<br>
- timestampSource = SOURCE_UNKNOWN
<br>
- whiteLevel = 1023
</p>

abdulhakam, 2022-11-07:
tried gcam, none work except one from b-s-g, and only 8.1.101..no newer or older versions, and only video mode works, no picures. but night mode on gcam go and "noise reduction" low light mode on open camera works fine., might be a problem with  accessing some hardware feature?

abdulhakam, 2022-11-07:
also tried rooting device, successfully unbricked it after bricking so maybe some privacy or "security" app might be the problem.

abdulhakam, 2023-02-28:
I see, I guess I should close it as solved since the problem is the device itself, thanks.

closed 2023-02-28 by abdulhakam

kaii-lb, 2024-05-03:
note that there is a raw module for pova devices, ld7 included

abdulhakam, 2024-05-05:
> note that there is a raw module for pova devices, ld7 included

Thank you for the reply. How do I enable/use it?

kaii-lb, 2024-05-05:
> > note that there is a raw module for pova devices, ld7 included
> 
> Thank you for the reply. How do I enable/use it?

https://disk.yandex.com/d/Zu2NRtwsvSHlrw/Tecno/Pova/Tecno%20Pova

download *one* of these andd install it with magisk. if it doesnt work try one of the others

abdulhakam, 2024-05-06:
well you see, I am unable to root the device, soft bricked many times in doing so! tried the same method on LD7J ( 16mp version), worked fine but not mine for some unknown reason.

kaii-lb, 2024-05-06:
> well you see, I am unable to root the device, soft bricked many times in doing so! tried the same method on LD7J ( 16mp version), worked fine but not mine for some unknown reason.

you probably need to flash empty vbmeta, but this is outside the scopee of this issue so lets not talk about it here.