eszdman/PhotonCamera #86: Crash on Pixel 6

Smorty100, 2022-11-19:
Expected behavior:
1. Open photoncamera
2. Take picture
3. Tap on image in the corner to view the image

Actual behavior:
1. Open photoncamera
2. Take picture (in normal photo mode)
3. App crashes once all frames for computation are captured
4. No image gets saved to gallery

Device: Google Pixel 6
OS: Android beta 13 (T1B3.221003.008)
Source: Downloaded from play store

ThatOneCalculator, 2023-01-30:
Same, but I get the crash right after the pic gets taken, before I even tap on the bottom left. P6 on GrapheneOS/Android 13.2

Zahrun, 2024-02-27:
On Fairphone 4, the app was crashing after photo is taken, and the photo was not saved.
I changed the Align Method from PCam CPU to PCam GPU, and now it is not crashing anymore.