eszdman/PhotonCamera #87: Save photos in AVIF or JPEG XL formats?

novomesk, 2023-01-24:
Hello,

I'd like to have possibility in PhotonCamera to save photos in new image formats like AVIF or JPEG XL.

These new formats have more efficient compression than old JPG, they can use more than 8bit/channel and they are also HDR ready.

AVIF can be supported via **libavif** and **libaom**,
JPEG XL can be saved via **libjxl**.

Please tell me if you agree or disagree with this feature.

Telegram desktop on Linux and Windows supports both AVIF and JXL now. I worked on enabling support in the Windows build of Telegram desktop app. My developer experiences on Android are limited but I'd like to help.

gianni-rosato, 2023-02-27:
I'll second this. Alternative codecs for exporting images would be great, as you can save space at low/medium quality with AVIF & save space at medium/high quality with JXL. Looking forward to your thoughts on this

nicolas-raoul, 2026-09-07:
JPEG XL format is an ISO open standard, royalty-free and with no patent issues.

This Android library can encode to JPEG XL: https://github.com/awxkee/jxl-coder
Example implementation: https://github.com/nicolas-raoul/OpenCamera-JPEG-XL

Thanks!