eszdman/PhotonCamera #89: [Device Support] Razer Phone 2

moriel5, 2023-02-11:
As it is, Photon Camera doesn't officially support the Razer Phone 2, and while it works on it, it is obvious that the quality is no different than using Footej Camera (I am using an old version from before it was bought out) or the stock camera.

There are modded version of Google Camera for the Razer Phone 2, in case it helps you: https://forum.xda-developers.com/t/mod-google-camera.3898230/

The Cameras are:

Rear
- Sony Exmor IMX363 (Main/Wide) - 12MP, f/1.75
- Sony Exmor IMX351 (Telephoto) - 12MP, f/2.6

Front
- Samsung 3H7 - 8MB, f/2.0

If any more information is needed, I'll be happy to produce it.
Likewise regarding testing.

eszdman, 2023-02-27:
I need a screenshot from about with device codename

moriel5, 2023-02-27:
I am sorry I had forgotten the device codename (and I kept thinking of the original Razer Phone's device, "cheryl", instead)

Razer Phone 2 codename: aura

moriel5, 2023-08-13:
I just found CameraIDs by one of the Photon Camera contributors, vibhorSRV, and I hope that the information that it exposes will be of help:

Sunday, August 13, 2023 1:15:57 PM GMT+03:00

============================

Device : razer Phone 2 (aura)
Manufacturer : Razer
Android : 9
Fingerprint : razer/cheryl2/aura:9/P-SMR3-RC002-RZR-200226/3141:user/release-keys

============================

Camera IDs Visible to Apps = [0, 1]

============================

All Camera IDs = [0, 1, 2, 3]

============================

CameraID = [0]  ★
Facing = BACK
Zoom = 1x
Type = MAIN
FocalLength = 4.29mm
35mm eqv FocalLength = 27.35mm
Aperture = 1.75
SensorSize = 5.6448x4.2336
PixelArray = 4032x3024
PixelSize = 1.40µm
AngleOfView(Diagonal) = 98°
AEModes = [0, 1, 2, 3, 4]
FlashSupported = true
RAW_SENSOR sizes = [4032x3024, 4032x2272]
SupportedHardwareLevel = FULL

============================

CameraID = [1]  ★
Facing = FRONT
Zoom = 1x
Type = MAIN
FocalLength = 3.51mm
35mm eqv FocalLength = 27.65mm
Aperture = 2.0
SensorSize = 4.5696x3.4272
PixelArray = 3264x2448
PixelSize = 1.40µm
AngleOfView(Diagonal) = 97°
AEModes = [0, 1]
FlashSupported = false
RAW_SENSOR sizes = [3264x2448, 3264x1836]
SupportedHardwareLevel = FULL

============================

CameraID = [2]  ★
Facing = BACK
Zoom = 1.34x
Type = DEPTH
FocalLength = 5.99mm
35mm eqv FocalLength = 36.60mm
Aperture = 2.6
SensorSize = 5.8912x4.368
PixelArray = 4208x3120
PixelSize = 1.40µm
AngleOfView(Diagonal) = 81°
AEModes = [0, 1]
FlashSupported = false
RAW_SENSOR sizes = [4208x3120, 4032x3024]
SupportedHardwareLevel = FULL

============================

CameraID = [3]
Facing = BACK
Type = LOGICAL
FocalLength = 4.29mm
35mm eqv FocalLength = 27.35mm
Aperture = 1.75
SensorSize = 5.6448x4.2336
PixelArray = 4032x3024
PixelSize = 1.40µm
AngleOfView(Diagonal) = 98°
AEModes = [0, 1, 2, 3, 4]
FlashSupported = true
RAW_SENSOR sizes = [4032x3024, 4032x2272, 4208x3120]
SupportedHardwareLevel = FULL

============================