eszdman/PhotonCamera #90: Why it use internet ?

iBabu007, 2023-02-16:
**Is your feature request related to a problem? Please describe.**

Why it use internet?
If possible then make it totally offline.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

eszdman, 2023-02-27:
It is necessary for the camera to be able to download the latest calibration data for each device from this github

rancidfrog, 2023-02-27:
@eszdman Is it possible to have the option to manually download it to proper location, instead of app using internet?

eszdman, 2023-02-27:
Not yet, but planned for future releases