eszdman/PhotonCamera #99: Device request: Pixel 6 pro

rawpie2, 2023-06-18:
Please support google:raven phones, it would be a great camera app over the google camera app! The app crashes when taking photos or switching cameras on the pixel.

chaseunc0, 2023-10-29:
Much milder issue on the pixel 6 (not the pro). App freezes while trying to use the wide angle lens, unfreezes when you switch back

kmaterzok, 2023-12-09:
The same issue has happened on Pixel 7 Pro:
* The choice of 1x lens have let me capture the image.
* 4.8x and 0.5x lens choices have not worked, just as in @chaseunc0's mention.

LunarPidgeon, 2025-09-26:
> The same issue has happened on Pixel 7 Pro:
> 
>     * The choice of 1x lens have let me capture the image.
> 
>     * 4.8x and 0.5x lens choices have not worked, just as in @chaseunc0's mention.

Experiencing this on pixel 7 pro as well. In my case, the app does not freeze, but the preview image does, on the last frame from the 1x lens. It is black if i select the 0.5x or 4.8x and then close the app, then reopen the app.

dafyddstephenson, 2025-12-16:
Throwing in another vote for the Pixel 6 Pro - having the same issue. App looks great, hope I can use it soon!