eszdman/PhotonCamera #106: Add device (e.g. blackview:bv9900pro)

gessel, 2024-03-21:
**Is your feature request related to a problem? Please describe.**

Adding support for a new device blackview:bv9900pro

**Describe the solution you'd like**

It'd be cool if there was a fairly straightforward way for users to generate the necessary specifications for additional devices, I assume things like the noise parameters and sensor IDs.  Is that something we can help with?

**Additional context**

For example, open camera reports about my device:

Main camera

Open Camera v1.52
Code: 88
Package: net.sourceforge.opencamera
Android API version: 29
Device manufacturer: Blackview
Device model: BV9900Pro
Device code-name: mt6779
Device variant: BV9900Pro
Language: en
Standard max heap?: 256
Large max heap?: 512
Display size: 1080x2060
Display metrics: 1080x2060
Current camera ID: 0
No. of cameras: 4
Multi-camera?: true
Camera API: Camera2 (Android L)
Camera orientation: 90
Photo mode: DRO
Min zoom factor: 1.0
Max zoom factor: 10.0
Preview resolutions: 1920x1080, 1440x1080, 1440x720, 1408x1056, 1280x960, 1280x720, 1080x1080, 960x960, 960x720, 960x540, 960x480, 800x600, 720x720, 720x480, 640x480, 352x288, 320x240, 192x144, 192x108, 176x144, 160x96
Preview resolution: 1440x1080
Photo resolutions: 4000x3000, 3000x3000, 3840x2160, 4000x2000, 3264x2448, 3072x1728, 2592x1944, 3168x1584, 2560x1920, 2304x1296, 1920x1440, 2176x1088, 1920x1088, 1920x1080, 1600x1200, 1440x1088, 1440x1080, 1408x1056, 1280x960, 1080x1080, 1440x720, 1280x720, 960x960, 960x720, 720x720, 800x600, 960x480, 640x480, 320x240
Photo resolution: 1280x960
Video qualities: 1, 6_r4000x2000, 6_r3072x1728, 6_r2592x1944, 6_r3168x1584, 6_r2560x1920, 6_r2560x1440, 6_r2560x1280, 6_r2304x1296, 6_r1920x1440, 6_r2176x1088, 6_r1920x1088, 6, 5_r1600x1200, 5_r1440x1088, 5_r1440x1080, 5_r1408x1056, 5_r1280x960, 5_r1080x1080, 5_r1440x720, 5, 5_r960x960, 4_r960x720, 4_r960x540, 4_r720x720, 4_r800x600, 4_r960x480, 4_r720x480, 4, 3, 7, 2_r192x144, 2, 0_r192x108, 0_r160x96
Video resolutions: 3840x2160, 4000x2000, 3072x1728, 2592x1944, 3168x1584, 2560x1920, 2560x1440, 2560x1280, 2304x1296, 1920x1440, 2176x1088, 1920x1088, 1920x1080, 1600x1200, 1440x1088, 1440x1080, 1408x1056, 1280x960, 1080x1080, 1440x720, 1280x720, 960x960, 960x720, 960x540, 720x720, 800x600, 960x480, 720x480, 640x480, 352x288, 320x240, 192x144, 176x144, 192x108, 160x96
Video quality: 6
Video frame width: 1920
Video frame height: 1080
Video bit rate: 20000000
Video frame rate: 30
Video capture rate: 30.0
Video high speed: false
Video capture rate factor: 1.0
Auto-level?: Available
Auto-level enabled?: false
Face detection?: Available
RAW?: Not available
HDR?: Available
Panorama?: Available
Gyro sensors?: Available
Expo?: Available
Expo compensation?: Available
Exposure compensation range: -4 to 4
Manual ISO?: Available
ISO range: 100 to 3200
Manual exposure?: Available
Exposure range: 100000 to 300000000
Manual WB?: Available
WB temperature: 1000 to 15000
Optical stabilization?: Not available
Optical stabilization enabled?: false
Video stabilization?: Available
Video stabilization enabled?: false
Tonemap curve?: Available
[... 344 lines cut ...]
Expo?: Available
Expo compensation?: Available
Exposure compensation range: -4 to 4
Manual ISO?: Not available
Manual exposure?: Not available
Manual WB?: Not available
Optical stabilization?: Not available
Optical stabilization enabled?: false
Video stabilization?: Available
Video stabilization enabled?: false
Tonemap curve?: Not available
Tonemap max curve points: 0
Can disable shutter sound?: Available
Camera view angle: 66.479 , 50.15833
Flash modes: flash_off, flash_frontscreen_on, flash_frontscreen_torch
Focus modes: None
Color effects: none, negative, sepia, whiteboard, blackboard, aqua
Scene modes: auto, action, portrait, landscape, night, night-portrait, theatre, beach, snow, sunset, steadyphoto, fireworks, sports, party, candlelight, barcode, hdr
White balances: auto, incandescent, fluorescent, warm-fluorescent, daylight, cloudy-daylight, twilight, shade
ISOs: None
Magnetic accuracy?: -1
Using SAF?: true
Save Location: OpenCamera
Save Location SAF: content://com.android.externalstorage.documents/tree/primary%3ADCIM%2FOpenCamera
Parameters: video-size=1920x1080;preferred-preview-size-for-video=1920x1080;preview-size-values=1920x1440,1920x1088,1920x1080,1600x1200,1440x1088,1440x1080,1440x720,1408x1056,1280x960,1280x720,960x960,960x720,960x480,800x600,720x720,720x480,640x480,352x288,320x240,192x144,192x108,176x144,160x96;video-size-values=2592x1944,2560x1280,2304x1296,2176x1088,2048x1536,1920x1440,1920x1088,1920x1080,1600x1200,1440x1088,1440x1080,1440x720,1408x1056,1280x960,1280x720,960x960,960x720,960x480,800x600,720x720,720x480,640x480,352x288,320x240,192x144,192x108,176x144,160x96;preview-format=yuv420sp;preview-format-values=yuv420p,yuv420sp,yuv420p;preview-fps-range-values=(10000,10000),(15000,15000),(15000,20000),(20000,20000),(5000,30000),(30000,30000);preview-frame-rate-values=10,15,20,30;picture-size-values=4608x3456,4608x2592,4608x2304,4160x3120,4096x2304,4032x3024,4032x2016,4000x3000,3840x2160,3456x3456,3264x2448,3168x1584,3104x3104,3072x1728,2592x1944,2560x1920,2560x1280,2304x1296,2176x1088,2048x1536,1920x1440,1920x1088,1920x1080,1600x1200,1440x1088,1440x1080,1440x720,1408x1056,1280x960,1280x720,960x960,960x720,960x480,800x600,720x720,640x480,320x240;picture-format=jpeg;picture-format-values=jpeg;jpeg-thumbnail-width=192;jpeg-thumbnail-height=144;jpeg-thumbnail-size-values=0x0,160x96,192x108,192x144;jpeg-thumbnail-quality=90;rotation=0;whitebalance=auto;whitebalance-values=auto,incandescent,fluorescent,warm-fluorescent,daylight,cloudy-daylight,twilight,shade;effect=none;effect-values=none,negative,sepia,whiteboard,blackboard,aqua;antibanding=auto;antibanding-values=off,50hz,60hz,auto;scene-mode=auto;scene-mode-values=auto,action,portrait,landscape,night,night-portrait,theatre,beach,snow,sunset,steadyphoto,fireworks,sports,party,candlelight,barcode,hdr;focus-mode=fixed;focus-mode-values=fixed;max-num-focus-areas=0;focus-areas=(0,0,0,0,0);focal-length=3.54;exposure-compensation=0;max-exposure-compensation=4;min-exposure-compensation=-4;exposure-compensation-step=0.5;auto-exposure-lock=false;auto-exposure-lock-supported=true;auto-whitebalance-lock=false;auto-whitebalance-lock-supported=true;max-num-metering-areas=1;max-zoom=99;zoom-ratios=100,103,106,109,112,115,118,121,124,127,130,133,136,139,142,145,148,151,154,157,160,163,166,169,172,175,178,181,184,187,190,193,196,199,203,206,209,212,215,218,221,224,227,230,233,236,239,242,245,248,251,254,257,260,263,266,269,272,275,278,281,284,287,290,293,296,299,303,306,309,312,315,318,321,324,327,330,333,336,339,342,345,348,351,354,357,360,363,366,369,372,375,378,381,384,387,390,393,396,399;zoom-supported=true;smooth-zoom-supported=false;focus-distances=Infinity,Infinity,Infinity;max-num-detected-faces-hw=15;max-num-detected-faces-sw=0;video-frame-format=android-opaque;video-stabilization-supported=true;video-snapshot-supported=false;video-stabilization=false;jpeg-quality=100;picture-size=4608x3456;preview-size=1440x1080;recording-hint=false;zoom=0;metering-areas=(0,0,0,0,0);preview-frame-rate=30;preview-fps-range=5000,30000;horizontal-view-angle=66.479;vertical-view-angle=52.2465

eszdman, 2024-05-12:
To check noise model and other configs you can create local file DCIM/PhotonCamera/Tuning/SensorSpecifics.txt containing information for your sensors
Examples: https://github.com/eszdman/PhotonCamera/tree/dev/app/specific/sensors
After selecting right noise models and other settings send me the file

eszdman, 2024-05-13:
ABCD means bayer pattern A,B,C,D
Here other noise models but in other format:
https://drive.google.com/drive/folders/18FBhvB7moD46Sn2QWpa9zZUN4X5rIYRBKKq_6NlufPwrgGQMNXkbNJA-OInawB64cO9U9zSJ