eszdman/PhotonCamera #108: Color temperature is not proper

daudarfin, 2024-05-09:
**Describe the bug**
Color temperature is not proper when I capture using the githup application. I don't see this issue with the Play store application.

Play store version works perfectly fine without any issues.

**To Reproduce**
Steps to reproduce the behavior:
1. Capture an image in Photo mode
2. See the color temperature

**Expected behavior**
Color temperature should be similar what we see in the camera preview

**Screenshots**
[
[image]
[image]
](url)

I could reproduce this issue on multiple android devices (including Google Pixel phones). Please note, my devices are not part of the supported list.

daudarfin, 2024-05-10:
I am using the latest source version (checked-out yesterday) and I don't see such method for color correction. Could you please let me know what is missing here? Any changes (to fix this issue) required in source code would also help.

eszdman, 2024-05-10:
Most likely this is due to the current session, I am using raw and jpeg stream at the same time
Some devices begin to behave differently when used this way at the moment I see the only solution is to enable dual session in device config file or upload config to repository

daudarfin, 2024-05-11:
Dual session will fix this issue?

Could you please let me know how to enable this session in the source code? Is it JPEG + RAW option available inside Settings?

eszdman, 2024-05-11:
> Could you please let me know how to enable this session in the source code?

https://github.com/eszdman/PhotonCamera/blob/e8f75501d2f7d1305feee7f590ff4c56eecd7d1a/app/src/main/java/com/particlesdevs/photoncamera/pro/Specific.java#L67

Crearte DeviceSpecific.txt in DCIM/PhotonCamera/tuning and write this line
isDualSessionSupported=true

daudarfin, 2024-05-13:
> Create DeviceSpecific.txt in DCIM/PhotonCamera/Tuning and write this line isDualSessionSupported=true

This doesn't help. Issue still persists even with dual session flag enabled.

eszdman, 2024-05-13:
Fixed crash in latest versions

daudarfin, 2024-05-14:
Thanks. The crash is fixed in the latest version.
 
But the color temperature issue still persists. The dual session support also didn't work. Could you please help me fixing this issue?

[image]
[image]