eszdman/PhotonCamera #109: Why does the app needs access to youtubei.goigleapis & dialercallinfolookup-pa

natarslan, 2024-05-09:
First of all thank you for developing this app. I had been comparing open camera, photon and my native camera for days now. I love the quality of the photos taken by Photon!

There is something that I dont understand (im not familiar with these topics). Im using netguard to follow apps and how they try to access internet. I can see so many different requests from photon every time I open the app (this is not the case for open camera nor in my native phone camera app).

In another [issue](https://github.com/eszdman/PhotonCamera/issues/90#issuecomment-1445966628) someone made it clear that the app needs access to internet to "download the latest calibration data for each device" from github . 

When i look at the internet access attempts in netguard I can see some with "raw.githubusercontent" which makes sense considering the above info. 

But i also see some attempts to:
- youtubei.googleapis.com
- dialercallinfolookup-pa.googleapis.com
- android.googleapis 
- ajax.googleapis
- And so on

Would it be possible for someone to explain  why these above requests are made in simple terms?

eszdman, 2024-05-09:
It depends on the version you are using, the version available on the playmarket uses firebase, current versions do not have this and only access github data

natarslan, 2024-05-10:
Thanks. I reinstalled the app by using the Google play link from the repo.

I no longer see access to youtubei.googleapis.com
dialercallinfolookup-pa.googleapis.com. 

Currently I only see access to 
- android.googleapis.com
- ajax.googleapis.com
- beacons.gvt2.com
- avatars.githubusercontent.com

And im assuming the first 3 happens because I downloaded the app from Google play. Is that right?

eszdman, 2024-05-10:
Install app from github releases

natarslan, 2024-05-10:
Yeap, thats what I was missing. Thank you for diricting me to the correct download.

closed 2024-05-10 by natarslan