eszdman/PhotonCamera #112: Interval Timer Shooting

Zig-Rust-Odin, 2024-05-21:
**Describe the solution you'd like**

* Select the shooting interval and number of shots Whether or not to enable the flash.
> https://helpguide.sony.net/ilc/1930/v1/en/contents/TP0002727486.html
Menu item details
Interval Shooting:
Sets whether to perform interval shooting. ([On]/[Off])
Shooting Start Time:
Sets the time from when you press the shutter button to when interval shooting starts. (1 second to 99 minutes 59 seconds)
Shooting Interval:
Sets the shooting interval (time from when one exposure starts to when the exposure for the next shot starts). (1 second to 60 seconds)
Number of Shots:
Sets the number of shots for interval shooting. (1 shot to 9999 shots)
AE Tracking Sensitivity:
Sets the tracking sensitivity of the auto exposure against the change in luminance during interval shooting. If you select [Low], exposure changes during interval shooting will be smoother. ([High]/[Mid]/[Low])
Silent Shoot. in Interval:
Sets whether or not to perform silent shooting during interval shooting. ([On]/[Off])
Shoot Interval Priority:
Sets whether or not to prioritize the shooting interval when the exposure mode is [Program Auto] or [Aperture Priority] and the shutter speed becomes longer than the time set for [Shooting Interval]. ([On]/[Off])

eszdman, 2024-05-21:
Already implemented in unlimited mode

Zig-Rust-Odin, 2024-05-22:
> Already implemented in unlimited mode

There's no time interval, there's no set number of shots, the flash isn't on.