eszdman/PhotonCamera #136: Device support request

brokoli-mate, 2025-03-07:
Please add support for Galaxy A23

ItzELECTR0, 2025-07-05:
Literally a hardware limitation they can't just "Add support" for it

brokoli-mate, 2025-07-22:
[image]

ItzELECTR0, 2025-07-22:
You mean 0.85? I have not used that version but the very latest uses a hardware permission for raw access which an A54 I used to own did NOT have. Not sure how you expect an A23 to have it.

It's not that they specifically dropped support for it, but they kinda did.

0.85 is really old and probably did not use that permission. I'd try it on that A54 I was talking about to prove my point but I don't own it anymore.